rjmcdougall/BB #6: No need to wait after Wire.requestFrom()

Koepel, 2017-08-15:
When the Wire.requestFrom() functions returns, the I2C transaction has completely finished and the received data is waiting in a buffer in the Wire library. There is no need to wait for something after the Wire.requestFrom().

The function "nonBlockingRead()" is therefor not needed. It is used in the files:
BB/arduino/BurnerBoardAndroidLight/BurnerBoardAndroidLight.ino
BB/arduino/BurnerBoardAzul/BurnerBoardAzul/BurnerBoardAzul.ino

If something was wrong on the I2C bus and the Wire.requestFrom() failed, then Wire.available() would probably return zero. In your code you don't even have to use Wire.available(), since you return -1 in case of an error, and the Wire.read() already returns -1 when there was no valid data and Wire.available() returns zero.

This does the same thing:
```
Wire.requestFrom(BQ34Z100,1);
return (Wire.read());
```

rjmcdougall, 2017-08-15:
thanks for your comment and thought.

I added this because I am getting hangs due to i2c. If there is packet loss on the wire interface the code to check the battery hangs. I'm not sure why yet.

It sounds like from what you are saying it cannot hand in requestFrom(). Could the send() hang?

Koepel, 2017-08-15:
The Wire library for AVR microcontroller is sadly not very good. That is an understatement. For an embedded system the Wire library is a terribly piece of code. It can halt in requestFrom() and never return. I was talking about a bus error that could be detected, sorry for not telling the whole story.

When there are long wires or wrong voltage levels, or when SDA or SCL is shortcut to GND or shortcut to each other, that could halt the requestFrom() and that will halt the sketch.
The rule of thumb is to have wires for SDA and SCL that are not longer than 50 cm; never put SDA and SCL next to each other in a flat ribbon cable; use a level shifter when connecting a 3.3V sensor to a 5V Arduino board; don't turn off the sensor; use Wire.setClock() to lower the clock speed; select the right value for the pullup resistors; and so on.

There is nothing you can do about that in software, your code will not help. The Wire.available() will only check the number of bytes in the buffer inside the Wire library.

If you are desperate, you might try a software implementation of the Wire library with timeouts. There are also many others that have altered the Arduino Wire library to add timeouts.

danielkeithw, 2019-05-28:
inactivity

closed 2019-05-28 by danielkeithw