rjmcdougall/BB #10: Server becomes unresponsive if bad video uploaded

rjmcdougall, 2018-05-07:
Burnerboard Brain goes whacky when a video is uploaded with bad content

danielkeithw, 2018-05-19:
also a bad audio.  Symptoms:

Power on brain and lights
watch battery pattern - NORMAL
colored matrix pattern - NORMAL
music - NORMAL
wait 60 seconds..  NORMAL
pattern 'stalls' for an instant
60 seconds
pattern repeatedly stalls
pattern freeze
music stops
30 seconds FROZEN
pattern restarts
music restarts
wait 60 seconds .. NORMAL
pattern freeze
music freeze
wait 30 seconds
pattern restart
music restart
the symptoms appear to be constant restarts of the brain software

This happened after i uploaded a 60BMP file.

danielkeithw, 2018-05-28:
my symptoms required a reflash of android.

danielkeithw, 2018-08-20:
this has not reproduced in quite some time.

closed 2018-08-20 by danielkeithw