rjmcdougall/BB #11: Add Friendly names for media tracks for voice and app

rjmcdougall, 2018-05-07:
It would be great if the app and board could have friendly names when it shows or announces tracks.

danielkeithw, 2018-06-17:
What i would like to do is have the MP3/MP4 themselves have friendly text in the name.  I can strip off the ".mp3"

Also I can change the algorithms to have friendly names.

danielkeithw, 2018-07-01:
I've added this for algorithms.  It is not exposed through the UI. This makes sense because I can't safely change the keys between the board/app for shared profiles on mixed versions.

 It seems to me that knowing if something is an MP3 or M4A is valuable because it tells us if we can sync or not. Friendly names remove that context.

commit 22296b6e (2018-07-01): Fixes #11
closed 2018-07-01 by danielkeithw