rjmcdougall/BB #48: App and Remote audio volumes are independent

rjmcdougall, 2018-05-14:
Woodson discovered that both audio volumes are independent for some reason

danielkeithw, 2018-05-28:
Bluetooth remote uses AudioManager which is at a lower level than the media player.  I am switching the Bluetooth LE to use AudioManager.

@rjmcdougall there are some volume control options for the radio which I am unsure how to test so I am leaving those alone.

closed 2018-05-28 by danielkeithw

danielkeithw, 2018-05-29:
> Dan; there's another factor here. Android has two volumes; one that is
> system wide and the other is the app. Because our remote control is a HID
> (human input device) then it changes the system volume outside of the app.
> I think we need to look into if we can fix the system vol rather than just
> the stream volume

The change I made Does change the system volume best I can tell.  (I verfiy
in android where the slider on the vysor session moves as I change things
via the app)

danielkeithw, 2018-05-30:
I double checked in the documentation and I do believe the change is correct:

danielkeithw, 2018-05-30:
@rjmcdougall I checked the android docs and i do believe this is the correct approach.  Here is the code snippet.

        AudioManager audioManager =
                (AudioManager)this.getSystemService(Context.AUDIO_SERVICE);

        int maxVolume = audioManager.getStreamMaxVolume(AudioManager.STREAM_MUSIC);
        int volume = audioManager.getStreamVolume(AudioManager.STREAM_MUSIC);
etc....

None of the other streams make sense for our scenario

STREAM_ACCESSIBILITYUsed to identify the volume of audio streams for accessibility prompts
int | STREAM_ALARMUsed to identify the volume of audio streams for alarms
int | STREAM_DTMFUsed to identify the volume of audio streams for DTMF Tones
int | STREAM_MUSICUsed to identify the volume of audio streams for music playback
int | STREAM_NOTIFICATIONUsed to identify the volume of audio streams for notifications
int | STREAM_RINGUsed to identify the volume of audio streams for the phone ring
int | STREAM_SYSTEMUsed to identify the volume of audio streams for system sounds
int | STREAM_VOICE_CALLUsed to identify the volume of audio streams for phone calls

Let me know if you see something I am missing.  My tests also verified it worked as expected (previous comment).  thanks!

commit 0a4983e8 (2018-05-28): Fixes #48