rjmcdougall/BB #67: Find my friends: Only keep/return board locations for 30 mins?

rjmcdougall, 2018-05-27:
Either this, or return the GPS age to the app, and let the app decide what to do with stale entries.

danielkeithw, 2018-06-03:
The app has minimal (nothing except board name) for persistence between sessions so i would like to return the age to the app.

edfine, 2018-06-08:
This could be visually encoded with color for quick scanning.

danielkeithw, 2018-07-01:
[image]

danielkeithw, 2018-07-01:
Return GMT datetime in minutes from board.
merge results with api response from cloud locations.
determine if boards last ack was over 1 hour ago and make it RED if so.

danielkeithw, 2018-07-01:
app is backwards compatible with previous BLE location responses.   Those bytes were unused (the altitude).

commit 3faf2f85 (2018-07-01): Fixes #67
closed 2018-07-01 by danielkeithw