rjmcdougall/BB #89: Why does the board keep repeating "Downloading new musice software: 97%"

edfine, 2018-06-10:
I'm not sure this is a bug, but it feels like a symptom of one.  Every few days it changes the percent.  The music available seems to change.  It never announces it is done.  

Steps:
Turn on Pegasus in range of a Burnerboard SSID.  

Expected:
Board says it is connecting.  then it announces it is downloading and the % increases until it announces it has completed downloading.

Observed:
It first says a high number like 97%, and every few minutes it announces the same percent.  Never finishes.  It does not noticeably affect audio tracks.

danielkeithw, 2018-06-10:
Need to understand what version of the software is running on Pegasus.  I work on adding that to the app.

I don't believe this bug is present in the current builds.

rjmcdougall, 2018-06-12:
Ed can you confirm -- I think this is gone now you are on a later build.

closed 2018-06-17 by rjmcdougall