rjmcdougall/BB #92: Audio Repeating and Throwing Errors

danielkeithw, 2018-06-16:
Latest Build from GH

Attempt to seek to past end of file: request = 678445190, durationMs = 7424369

repeated errors.

+ the track repeats its first few seconds over and over, never advancing.

rjmcdougall, 2018-06-16:
Can you describe which profile this is happening? Can you also cut and paste the json from the download URL?

rjmcdougall, 2018-06-16:
I think this means the download partially failed, but the file is there. Maybe the file-length check still isn't working, so the downloader thinks the file is ok, but it's only a partial download?

rjmcdougall, 2018-06-16:
One way to check: adb shell into device.

Look in /data/data/com.richardmcdougall.bb/files for the file it's playing and compare it to the download size.

danielkeithw, 2018-06-16:
{
    "audio": [
        {
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/test.m4a",
            "ordinal": 1,
            "Size": 985180,
            "Length": 304,
            "localName": "test.m4a"
        },
        {
            "localName": "02-BombingRun2.mp3",
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/02-BombingRun2.mp3",
            "ordinal": 2,
            "Size": 118796042,
            "Length": 738044
        },
        {
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/01-FunRide2.mp3",
            "ordinal": 3,
            "Size": 122529253,
            "Length": 762037,
            "localName": "01-FunRide2.mp3"
        },
        {
            "Size": 196019403,
            "Length": 1224011,
            "localName": "test 305.mp3",
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/test 305.mp3",
            "ordinal": 4
        }
    ],
    "video": [
        {
            "algorithm": "modeMatrix(kMatrixBurnerColor)",
            "ordinal": 0
        },
        {
            "ordinal": 1,
            "algorithm": "modeAudioCenter()"
        },
        {
            "ordinal": 2,
            "Size": 16356811,
            "localName": "jellyfish.mp4",
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/jellyfish.mp4"
        },
        {
            "ordinal": 3,
            "Size": 1362949,
            "localName": "wavey_colors.mp4",
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/wavey_colors.mp4"
        },
        {
            "localName": "worms.mp4",
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/worms.mp4",
            "ordinal": 4,
            "Size": 1113837,
            "SpeachCue": "",
            "Length": 30
        },
        {
            "localName": "Tunnels-2.mp4",
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/Tunnels-2.mp4",
            "ordinal": 6,
            "Size": 70608093,
            "SpeachCue": "",
            "Length": 3144
        },
        {
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/Red Balloons.mp4",
            "ordinal": 7,
            "Size": 247560,
            "SpeachCue": "",
            "Length": 49,
            "localName": "Red Balloons.mp4"
        },
        {
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/tetris.mp4",
            "ordinal": 8,
            "Size": 13757050,
            "SpeachCue": "",
            "Length": 704,
            "localName": "tetris.mp4"
        },
        {
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/StanfordLogo.mp4",
            "ordinal": 9,
            "Size": 8502786,
            "SpeachCue": "",
            "Length": 1396,
            "localName": "StanfordLogo.mp4"
        },
        {
            "localName": "60 BPM.mp4",
            "URL": "https://storage.googleapis.com/burner-board/BurnerBoardMedia/sexy/testsexy/60 BPM.mp4",
            "ordinal": 10,
            "Size": 985180,
            "SpeachCue": "",
            "Length": 304
        }
    ]
}

danielkeithw, 2018-06-16:
Does NOT happen with test.m4a. Does with other tracks.

Does happen when remotely debugging.
Does happen when not debugging but logcat-ing.

However, if I reboot sexy and do not ADB to it at all, this does NOT happen.

So, I think we will be fine in production.  I'd still like to know what is going on...

danielkeithw, 2018-06-24:
Does not happen w/ latest release.

closed 2018-06-24 by danielkeithw