rjmcdougall/BB #94: Redeploy burnerboard.com as node.js standard appengine

rjmcdougall, 2018-06-20:
Google just released native node support for appengine. 

Lets redeploy the app using the new appengine type -- it should be 1/10th of the running cost.

rjmcdougall, 2018-06-20:
https://cloud.google.com/nodejs/

danielkeithw, 2018-07-25:
errors when importing any reasonably sized m4a file. smaller mpegs worked fine

<html><head> <meta http-equiv="content-type" content="text/html;charset=utf-8"> <title>500 Server Error</title> </head> <body text=#000000 bgcolor=#ffffff> <h1>Error: Server Error</h1> <h2>The server encountered an error and could not complete your request.<p>Please try again in 30 seconds.</h2> <h2></h2> </body></html>

rjmcdougall, 2018-07-25:
> <title>500 Server Error</title> Error: Server Error The server

Bugger. How did you figure that before?

danielkeithw, 2019-01-13:
Two changes:
needed instances with up to 1G of memory due to FFMPEG.
Node version support issues.  Needed Node10 engine.  Now beta but works...

commit 052c44cb (2019-01-13): Fixes #94
closed 2019-01-13 by danielkeithw