rjmcdougall/BB #107: Visualization service crashing / race condition.

danielkeithw, 2018-06-27:
With the latest build...       

mode is 0 (after += -1).  GetVideo(mode) is returning null.   The .has("algorithm") is throwing exceptions.  This allows the code to proceed, but I'm not sure what the underlying issue is.

 // TODO: check perf overhead of checking this every frame
        if(mBurnerBoard.mBBService.dlManager.GetVideo(mode) != null) {
            if (mBurnerBoard.mBBService.dlManager.GetVideo(mode).has("algorithm")) {
                String algorithm = mBurnerBoard.mBBService.dlManager.GetAlgorithm(mode);
                return displayAlgorithm(algorithm);
            } else {
                mVisualizationVideo.update(mode);
                return mFrameRate;
            }
        }
        else
            return mFrameRate;

danielkeithw, 2018-06-28:
if(mBurnerBoard.mBBService.dlManager.dataDirectory == null){
            return mFrameRate;
        }

This happens if the directory is not loaded when this method gets hit. fix is return if the directory is null

rjmcdougall, 2018-06-28:
Crashed on Cranky with blank profile after fix.

commit f55f7a2a (2018-06-28): Fixes #107
closed 2018-06-28 by rjmcdougall