rjmcdougall/BB #120: Check if we can provision a new brain on the playa with no cloud

rjmcdougall, 2018-07-01:
We won't have a cloud to sync profiles and board settings from.

It should be feasible to provision a new brain if we have a static copy of the files that we can copy in place into the /files directory.

This placeholder is to check that we can do that.

rjmcdougall, 2018-07-06:
@danielkeithw do we need to make a stand-alone service? Likely a lot of work because of GCS etc.

danielkeithw, 2018-07-06:
If we copy all of the data/data/files from any board onto any other it will work.  We won't need a service.  I'll test this in the weeks leading up to the playa to be sure.  There are two files:

board.json contains ALL of the metadata for the boards.
directory.json contains ALL of the metadata for the active profile on the board.  

None of them have any need to phone home to the cloud.  

Also, the app doesn't have any need to phone home to the cloud.  The app has a copy of boards.json but it can survive without one altogether. It only uses it to show friendly names and also the board-specific colors.

closed 2019-01-02 by danielkeithw