rjmcdougall/BB #135: Music doesn't sync in build 3910.

rjmcdougall, 2018-07-09:


danielkeithw, 2018-07-09:
ctrl_interface=/data/misc/wifi/sockets
disable_scan_offload=1
driver_param=use_p2p_group_interface=1
update_config=1
device_name=msm8916_64
manufacturer=Burner Board Azul
model_name=vega
model_number=vega
serial_number=bd602528
device_type=10-0050F204-5
config_methods=physical_display virtual_push_button
p2p_disabled=1
p2p_no_group_iface=1
pmf=1
external_sim=1

network={
	ssid="ATT7Cd2QyD"
	psk="79cj+rbw%e96"
	key_mgmt=WPA-PSK
	priority=1
}

network={
	ssid="burnerboard"
	psk="firetruck"
}

network={
	ssid="burnerboard"
	psk="firetruck"
	disabled=1
}

network={
	ssid="burnerboard"
	psk="firetruck"
	disabled=1
}

network={
	ssid="burnerboard"
	psk="firetruck"
	disabled=1
}

network={
	ssid="burnerboard"
	psk="firetruck"
}

danielkeithw, 2018-07-09:
Looks like Vega can control the tracks on sexy but not vice-versa.

Both have master running.

danielkeithw, 2018-07-09:
All appears to be working.