rjmcdougall/BB #145: Charge my iPhone via board

danielkeithw, 2018-07-15:
run an iphone cable from a USB port on the brain to prevent emergecy power outage!

rjmcdougall, 2018-07-16:
there are no spare USB ports on the brain now. We could add a simple USB charger:

https://www.amazon.com/SonRu-Charger-Socket-Waterproof-Motorcycle/dp/B07CG52DTB/ref=sr_1_5?ie=UTF8&qid=1531721343&sr=8-5&keywords=12v+usb+wired+charger

woodsonmartin, 2018-07-18:
Why not embed a cable on a spring loaded coil so you can just tug on a charger cable and plug in your phone?

danielkeithw, 2018-07-18:
Yes!!!!

woodsonmartin, 2018-07-20:
This could be a useful part of the proposed system: 

https://www.amazon.com/DecoStain-Adapter-Waterproof-Motorcycle-Charger/dp/B073VCL8Z2/ref=sr_1_14?ie=UTF8&qid=1532110132&sr=8-14&keywords=dc+powered+usb+charger

woodsonmartin, 2018-07-20:
Maybe this for the retractable charging cable: https://www.target.com/p/gems-174-micro-usb-pod-cable-management-north-white/-/A-52314120?ref=tgt_adv_XS000000&AFID=google_pla_df&CPNG=PLA_Electronics+Shopping&adgroup=SC_Electronics&LID=700000001170770pgs&network=g&device=c&location=9031914&gclid=CjwKCAjwkMbaBRBAEiwAlH5v_kNtAYhnMNHtzWTS2SeAECI63h-w2jzqPT7QroOIQNO0LhzwXamWSBoCARsQAvD_BwE&gclsrc=aw.ds

danielkeithw, 2018-08-20:
I have a working one on Vega now.  Could be improved.

closed 2018-08-20 by danielkeithw