rjmcdougall/BB #169: Download manager keeps trying to download non-existing files

jib, 2018-08-15:
```
08-15 17:50:39.787  1956  2023 D BB.BBDownloadManger: Downloaded JSON: {"audio":[{"localName":"distrikt_2013-04-03T07_00_00-07_00.mp3","URL":"https://storage.googleapis.com/burner-board/BurnerBoardMedia/global/menlomickey/distrikt_2013-04-03T07_00_00-07_00.mp3","ordinal":1,"Size":166758533,"Length":4159}],"video":[{"algorithm":"MickeyGold()","ordinal":0},{"ordinal":0,"algorithm":"modeMickeyBlank()"},{"algorithm":"MickeyBlueGold()","ordinal":0},{"algorithm":"modeFire(kModeFireNormal)","ordinal":0,"Size":0},{"algorithm":"modeMatrix(kMatrixReverse)","ordinal":0},{"algorithm":"MickeySparkle()","ordinal":1},{"algorithm":"MickeyColors()","ordinal":2},{"URL":"https://storage.googleapis.com/burner-board/BurnerBoardMedia/vega/Vega1/lightrays.mp4","ordinal":3,"Size":10654549,"localName":"lightrays.mp4"},{"URL":"https://storage.googleapis.com/burner-board/BurnerBoardMedia/vega/Vega1/repeat_fire.mp4","ordinal":4,"Size":1033290,"localName":"repeat_fire.mp4"},{"localName":"disco_lights.mp4","URL":"https://storage.googleapis.com/burner-board/BurnerBoardMedia/vega/Vega1/disco_lights.mp4","ordinal":5,"Size":2124367}]}
08-15 17:50:40.338  1956  2023 W System.err: java.io.FileNotFoundException: https://storage.googleapis.com/burner-board/BurnerBoardMedia/vega/Vega1/lightrays.mp4
08-15 17:50:40.338  1956  2023 W System.err: 	at com.android.okhttp.internal.huc.HttpURLConnectionImpl.getInputStream(HttpURLConnectionImpl.java:251)
08-15 17:50:40.338  1956  2023 W System.err: 	at com.android.okhttp.internal.huc.DelegatingHttpsURLConnection.getInputStream(DelegatingHttpsURLConnection.java:210)
08-15 17:50:40.338  1956  2023 W System.err: 	at com.android.okhttp.internal.huc.HttpsURLConnectionImpl.getInputStream(Unknown Source:0)
08-15 17:50:40.338  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.DownloadURL(DownloadManager.java:255)
08-15 17:50:40.339  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.replaceFile(DownloadManager.java:406)
08-15 17:50:40.339  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.GetNewDirectory(DownloadManager.java:473)
08-15 17:50:40.339  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.run(DownloadManager.java:575)
08-15 17:50:40.339  1956  2023 W System.err: 	at java.lang.Thread.run(Thread.java:764)
08-15 17:50:40.482  1956  1956 D mBBEventReciever: onReceive entered
08-15 17:50:40.532  1956  2023 W System.err: java.io.FileNotFoundException: https://storage.googleapis.com/burner-board/BurnerBoardMedia/vega/Vega1/repeat_fire.mp4
08-15 17:50:40.533  1956  2023 W System.err: 	at com.android.okhttp.internal.huc.HttpURLConnectionImpl.getInputStream(HttpURLConnectionImpl.java:251)
08-15 17:50:40.533  1956  2023 W System.err: 	at com.android.okhttp.internal.huc.DelegatingHttpsURLConnection.getInputStream(DelegatingHttpsURLConnection.java:210)
08-15 17:50:40.533  1956  2023 W System.err: 	at com.android.okhttp.internal.huc.HttpsURLConnectionImpl.getInputStream(Unknown Source:0)
08-15 17:50:40.533  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.DownloadURL(DownloadManager.java:255)
08-15 17:50:40.533  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.replaceFile(DownloadManager.java:406)
08-15 17:50:40.533  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.GetNewDirectory(DownloadManager.java:473)
08-15 17:50:40.533  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.run(DownloadManager.java:575)
08-15 17:50:40.533  1956  2023 W System.err: 	at java.lang.Thread.run(Thread.java:764)
08-15 17:50:40.557  1956  1956 V BB.BluetoothRemote: adding new unpaired device: 00:0D:4B:CC:1F:BE-00:0D:4B:CC:1F:BE
08-15 17:50:40.799  1956  2027 V BB.BoardVisualization: Frames: 100
08-15 17:50:40.963  1956  2023 W System.err: java.io.FileNotFoundException: https://storage.googleapis.com/burner-board/BurnerBoardMedia/vega/Vega1/disco_lights.mp4
08-15 17:50:40.963  1956  2023 W System.err: 	at com.android.okhttp.internal.huc.HttpURLConnectionImpl.getInputStream(HttpURLConnectionImpl.java:251)
08-15 17:50:40.963  1956  2023 W System.err: 	at com.android.okhttp.internal.huc.DelegatingHttpsURLConnection.getInputStream(DelegatingHttpsURLConnection.java:210)
08-15 17:50:40.963  1956  2023 W System.err: 	at com.android.okhttp.internal.huc.HttpsURLConnectionImpl.getInputStream(Unknown Source:0)
08-15 17:50:40.963  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.DownloadURL(DownloadManager.java:255)
08-15 17:50:40.963  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.replaceFile(DownloadManager.java:406)
08-15 17:50:40.963  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.GetNewDirectory(DownloadManager.java:473)
08-15 17:50:40.963  1956  2023 W System.err: 	at com.richardmcdougall.bb.DownloadManager$BackgroundThread.run(DownloadManager.java:575)
08-15 17:50:40.963  1956  2023 W System.err: 	at java.lang.Thread.run(Thread.java:764)
```

jib, 2018-08-15:
This happens once a minute, including the voice announcemens that claim 'success'.

To make matters worse, I can't remove these items from the UI as they're no longer in the database (opening second ticket)

jib, 2018-08-15:
Scratch that last comment - the UI is misleading; it throws an error about missing object, but it DOES remove it from the profile

danielkeithw, 2018-08-20:
Each file exists in one and only one directory by design.  Each profile has its own copies.

closed 2018-08-20 by danielkeithw