rjmcdougall/BB #210: JosPack 2018 causes Azul boards to stop responding to the app via bluetooth

edfine, 2019-07-23:
Simply turning on a pack in bluetooth range of an Azul board causes the board to stop being available to the app.  Music and lights continue.  Turning the pack off and rebooting the brain allows it to be discoverable again.  
I have not tested the Satechi.  

Steps to reproduce: 
Turn on Azul board (e.g. Pegasus) and connect with app.
Turn on un-updated JosPack (Statler) with the build from 2018 burn.

Expected: 
Be able to see the Azul board with the app and connect with no effect from the JosPack.
Observed:  
Note that the board is
[image]
[image]
[image]
[image]
[image]

 no longer connected and the board is no longer discoverable.

edfine, 2019-07-23:
@danielkeithw @rjmcdougall This is the JosPack bug.  
Perhaps we could work around by updating all JosPacks, but it may be better to figure out what's up with the Azul Brain.

danielkeithw, 2019-07-23:
I think it may be with the app and not the azul Brain. Thank you I’ll take
a look!

danielkeithw, 2019-10-13:
deprecating rpi

closed 2019-10-13 by danielkeithw